jglathe/linux_ms_dev_kit #22: Help wanted not same device related

Saikatsaha1996, 2024-10-19:
Hi ! 
I am Saikat Saha I am porting my downstream device to mainline.. I am facing some issues.. which you faced already.. 
My issue 

```
[    1.151192] dwc3 a600000.usb: Adding to iommu group 8
[    1.207374] qcom_pmic_glink pmic-glink: Failed to create device link (0x180) with a600000.usb
[    2.611013] synth uevent: /devices/platform/pmic-glink/pmic_glink.power-supply.0/power_supply/qcom-battmgr-bat: failed to send uevent
[    2.611228] power_supply qcom-battmgr-bat: uevent: failed to send synthetic uevent: -11
[    2.611328] synth uevent: /devices/platform/pmic-glink/pmic_glink.power-supply.0/power_supply/qcom-battmgr-usb: failed to send uevent
[    2.611347] power_supply qcom-battmgr-usb: uevent: failed to send synthetic uevent: -11
[    2.611410] synth uevent: /devices/platform/pmic-glink/pmic_glink.power-supply.0/power_supply/qcom-battmgr-wls: failed to send uevent
[    2.611428] power_supply qcom-battmgr-wls: uevent: failed to send synthetic uevent: -11
[    2.729598] qcom_pmic_glink pmic-glink: Failed to create device link (0x180) with 0-0042
```
I want to know how you fix it ? Please share some suggestions if possible..

jglathe, 2024-10-19:
Hi! Well those pmic-glink errors are quite persistent and I have no idea how to fix them. 

[image]

[image]

On the WDK it doesn't really affect anything (that I notice). On the HP Omnibook X14 (x1e laptop) I get exactly the same errors, but battery life is good and suspend / resume works. :man_shrugging:

Saikatsaha1996, 2024-10-19:
> Hi! Well those pmic-glink errors are quite persistent and I have no idea how to fix them. [...]

Thank you so much for your response..

I actually have charging issue for my device.. so it can be reason for charging?

jglathe, 2024-10-19:
This is something else. Looks like the adsp firmware is not loaded.

Saikatsaha1996, 2024-10-19:
> This is something else. Looks like the adsp firmware is not loaded.

Adsp loaded and booted up successfully 
[armbian3.txt](https://github.com/user-attachments/files/17447304/armbian3.txt)

jglathe, 2024-10-19:
hmm do you have the pd-mapper on in the kernel , or do you use pd-mapper and qrtr as user mode processes? Also, if there are defining jsn files, are they available (at the same place as the firmware) and loaded?

Saikatsaha1996, 2024-10-19:
> hmm do you have the pd-mapper on in the kernel , or do you use pd-mapper and qrtr as user mode processes? Also, if there are defining jsn files, are they available (at the same place as the firmware) and loaded?

Yea.. pd-mapper available..
https://github.com/Saikatsaha1996/build/blob/a4621a4808ff53b0104bee2ec3206dc4fcbcc415/config/boards/xiaomi-elish.conf#L76

And jsn file also available all firmwares
https://github.com/Saikatsaha1996/firmware/tree/porsche/qcom/sm8350/realme/porsche

Saikatsaha1996, 2024-10-19:
> hmm do you have the pd-mapper on in the kernel , or do you use pd-mapper and qrtr as user mode processes? Also, if there are defining jsn files, are they available (at the same place as the firmware) and loaded?

Okay most probably charger_pd adsp.jsn file is missing?
https://github.com/Saikatsaha1996/android_kernel_oneplus_sm8350/blob/9c86b0cf0172c8ca42ed377af48c0e877ba7a050/arch/arm64/boot/dts/vendor/qcom/lahaina.dtsi#L5591

Saikatsaha1996, 2024-10-19:
> hmm do you have the pd-mapper on in the kernel , or do you use pd-mapper and qrtr as user mode processes? Also, if there are defining jsn files, are they available (at the same place as the firmware) and loaded?

Okay got it this one right?
https://github.com/Saikatsaha1996/firmware-realme-porsche/blob/porsche/lib/firmware/qcom/sm8350/porsche/battmgr.jsn

Saikatsaha1996, 2024-10-20:
Finally qcom battmgr working.. with `battmgr.jsn` but still device not charging..
[image]

jglathe, 2024-10-20:
are there other .jsn files? I don't know enough to assist in debugging it deeper.

Saikatsaha1996, 2024-10-20:
> are there other .jsn files? I don't know enough to assist in debugging it deeper.

But magically you give me a strong client for fix my battery manager.. thank you so much 😁..

This is my total full firmwares.. no more jsn file available
https://github.com/Saikatsaha1996/firmware-realme-porsche/tree/porsche/lib/firmware/qcom/sm8350/porsche

jglathe, 2024-10-20:
Hmm looks to me that there are other .jsn files (like adspr.jsn). For sc8280xp it is usually four files, but yours is sm8350:

[image]

Those 4 files are in the firmware directory, too. :man_shrugging: I would recommend to have them in the initramfs, like [here](https://github.com/jglathe/wdk2023_syshacks/blob/jg/hp-omnibook-x14/etc/initramfs-tools/hooks/x14-firmware) example only, need to do this with your paths and firmwares). What I can tell (from binwalking UEFI updates) is that it appears to be usually these 4 .jsn files you need.

Saikatsaha1996, 2024-10-20:
> Hmm looks to me that there are other .jsn files (like adspr.jsn). For sc8280xp it is usually four files, but yours is sm8350:
> 
> ![image](https://private-user-images.githubusercontent.com/33185912/378190062-70339407-f11b-44e2-bc13-3202811ec102.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0Njg0MzksIm5iZiI6MTcyOTQ2ODEzOSwicGF0aCI6Ii8zMzE4NTkxMi8zNzgxOTAwNjItNzAzMzk0MDctZjExYi00NGUyLWJjMTMtMzIwMjgxMWVjMTAyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIwVDIzNDg1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM2MmFlM2YwMTA4YzIyNGNjNjM1MjgyOTZkZDE5MWM2OGJlNzlkZTY0OWY1MGU2NmNmZjRiYWIwZDk5YWJhODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.vKB7b4kt3lkSQ7LSLUX0cMGEIhpEYZmlNL5ERcfDGtg)
> 
> Those 4 files are in the firmware directory, too. 🤷‍♂️ I would recommend to have them in the initramfs, like [here](https://github.com/jglathe/wdk2023_syshacks/blob/jg/hp-omnibook-x14/etc/initramfs-tools/hooks/x14-firmware) example only, need to do this with your paths and firmwares). What I can tell (from binwalking UEFI updates) is that it appears to be usually these 4 .jsn files you need.

I already all does have.. like

```
https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/adspr.jsn
https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/adspua.jsn
https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/battmgr.jsn
https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/cdspr.jsn
https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/modemr.jsn
https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/slpir.jsn
https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/slpius.jsn
```

Saikatsaha1996, 2024-10-21:
> I already all does have.. like
>
> ```
> https://github.com/Saikatsaha1996/firmware/blob/porsche/qcom/sm8350/realme/porsche/adspr.jsn [...]

Finally solved my device charging issue.. now device charging at full speed.. but still one issue present.. like battery charging indicator not showing.. only battery persentage increased & dmesg outputs..
[image]
[image]

jglathe, 2024-10-21:
That is way more talkative than the X13s :laughing:  Honestly, no idea.

closed 2024-10-27 by jglathe