jglathe/linux_ms_dev_kit #24: Ubuntu-Concept [X1E] Support HP Omnibook X 14

Insane-Redditor, 2024-10-20:
Transferred over here from Ubuntu website. As said earlier, the image doesn't boot. When I go to select it via Ventoy, it doesn't give me the normal options of booting normally or via grub. When flashed to a usb, it does a similar thing. It attempts to boot, because it shows the HP screen of numbers + enter to confirm disabling secure boot. But it constantly does that until I power the device down. 

With the official ubuntu concept image, via Ventoy and flashed to a usb, it booted to the grub menu before loading for a second and then rebooting back to the numbers + enter to confirm disabling secure boot. 

The image I used was the one you told me to try: Ubuntu_Desktop_24.04_x1e_6.11rc_v6.img

I see that there are "newer" ones denoted by higher v number, so I'll try the Ubuntu_Desktop_24.04_x1e_6.11rc_v7.img and then report back again.

smedir, 2024-10-20:
Look in discussions, there are some tips there as well. Version 7 does seem to boot on lenovo slim 7x and ubuntu comes up.

jglathe, 2024-10-20:
Hmm just fighting with v8 and a new contender (Snapdragon Dev Kit for Windows, codenamed "angry box" 'cause it sounds like it)... You need to get into the boot manager of the UEFI. Usually, after disabling secure boot in UEFI (and hopefully after decrypting the Windows drive), it asks only once for confirmation that it disables secure boot. Afterwards you need to power on an press F9 repeatedly (and have the stick / ssd plugged in ofcourse). The Boot device selection should come up. You select the external disk, hopefully get the grub menu, hopefully can select the HP X14, and off it goes. If Keyboard doesn't work (it happens), try using an external KB. 

[image]

jglathe, 2024-10-20:
Oh and one other caveat: Seems to be a good idea to have a USB to Ethernet adapter plugged in to ensure ethernet connectivity. Cloud-Init seems to get angry when no network is available. The default board file on the image doesn't have the entry required for this laptop (yet).

Insane-Redditor, 2024-10-20:
Well, after much keyboard spamming, this is all I get. [image]

jglathe, 2024-10-20:
Let me guess: You're booting from USB Type-A? This could be still not working (didn't come around to try, although it should). Boot from Type-C works. The other potential issue is no network available. It is what I saw when working on v8, using an USB ethernet adapter and a type-c SSD. I would bet my money on the ethernet connectivity, though.

[image]

Another thing I noticed and will be hopefully fixed in v8: ntp timesync got borked in the default setup, so date and time needs to be set manually. This is bad because date and time need to be halfway accurate for apt updates to work.

Insane-Redditor, 2024-10-20:
I booted using a type c dongle. It took a bunch of random keyboard presses until I was able to select the Omnibook to boot. Once I got to the Linux desktop, it worked, but it was quite buggy and slow. I haven't installed it yet. It didn't give me an option to select a drive or partition, so I didn't install it. [image]

jglathe, 2024-10-20:
Okay it didn't install. But it booted up to user space :grin: Random key presses? Curious about that. 
Regarding what is to expect as of now: That it boots to user space and gets you a GUI desktop for further tinkering :smirk: Please see [wiki (under construction)](https://github.com/jglathe/linux_ms_dev_kit/wiki/Bootable-Image-for-multiple-Snapdragon-(SC8280XP)-and-Snapdragon-X-Elite-(X1E80100)-laptops) It is not an installer ISO like the [Ubuntu-concept X1E](https://discourse.ubuntu.com/t/ubuntu-24-10-concept-snapdragon-x-elite/48800/1), it is pre-configured desktop like on a Raspberry Pi. 

You could do a lot of improvements after the first bootup, though. 

- [fetch the firmwares](https://github.com/jglathe/wdk2023_fw_fetch/blob/jg/hp-omnibook-x14/bin/fetch_x1e80100_fw.sh) for the x1e SoC from the Windows Disk - this gives you accelerated graphics (script is on v8)
- get ntp timesync to run again (licked in v8)
- update grub to also boot Windows
- [get the right board-2.bin](https://github.com/qca/qca-swiss-army-knife/pull/10) for working WLAN (should do in V8)
- [get the right firmware for Bluetooth](https://github.com/jglathe/linux_ms_dev_kit/wiki/Getting-better-range-and-quality-from-the-WCN6855-Bluetooth-chipset)
- upgrade to 24.10
- install on an SSD - although I'd recommend to use the official Ubuntu ISO when it works.

closed 2024-10-20 by jglathe

Insane-Redditor, 2024-10-21:
> Okay it didn't install. But it booted up to user space 😁 Random key presses? Curious about that. Regarding what is to expect as of now: That it boots to user space and gets you a GUI desktop for further tinkering 😏 Please see [wiki (under construction)](https://github.com/jglathe/linux_ms_dev_kit/wiki/Bootable-Image-for-multiple-Snapdragon-(SC8280XP)-and-Snapdragon-X-Elite-(X1E80100)-laptops) It is not an installer ISO like the [Ubuntu-concept X1E](https://discourse.ubuntu.com/t/ubuntu-24-10-concept-snapdragon-x-elite/48800/1), it is pre-configured desktop like on a Raspberry Pi.
> 
> You could do a lot of improvements after the first bootup, though.
> 
> * [fetch the firmwares](https://github.com/jglathe/wdk2023_fw_fetch/blob/jg/hp-omnibook-x14/bin/fetch_x1e80100_fw.sh) for the x1e SoC from the Windows Disk - this gives you accelerated graphics (script is on v8)
> * get ntp timesync to run again (licked in v8)
> * update grub to also boot Windows
> * [get the right board-2.bin](https://github.com/qca/qca-swiss-army-knife/pull/10) for working WLAN (should do in V8)
> * [get the right firmware for Bluetooth](https://github.com/jglathe/linux_ms_dev_kit/wiki/Getting-better-range-and-quality-from-the-WCN6855-Bluetooth-chipset)
> * upgrade to 24.10
> * install on an SSD - although I'd recommend to use the official Ubuntu ISO when it works.

Ah, okay, that explains why I couldn't install it. Thank you.