jglathe/linux_ms_dev_kit #26: SnapDragon 8380 DK bootup issue

Vancezheng, 2024-11-05:
We created a bootup disk using Ubuntu_Desktop_24.04_x1e001dev_ssh_6.12.img.xz and transferred the file system to the 8380DK. After booting, the kernel log can be printed to the HDMI screen, but after 0.6 seconds, the kernel log switches from the HDMI screen to the serial terminal output, and the HDMI screen remains black. However, the Ubuntu system can start normally. After modifying the kernel cmdline to console=fbcon, the HDMI screen continuously prints the kernel log, but a kernel panic occurs when it reaches the init process. I am not sure why this happens. I want to boot into text mode or GUI. What should I do？ Thanks.

https://github.com/user-attachments/assets/319b50ac-6ec4-46f8-8fc7-47ec2b630c0b

[image]

jglathe, 2024-11-05:
Hmm Did you add the firmwares? It looks like some info goes missing, seen this with the wrong (?) dtb. You're booting from local SSD, how is boot wired up? Just a dd copy of the image to the SSD?

jglathe, 2024-11-05:
interesting that fbcon should do the trick though, thanks. I'll have time to experiment i a few days, now is work scheduled

Vancezheng, 2024-11-05:
> Hmm Did you add the firmwares? It looks like some info goes missing, seen this with the wrong (?) dtb. You're booting from local SSD, how is boot wired up? Just a dd copy of the image to the SSD?

We used cp to copy the ubuntu file system from the USB disk to the SSD disk, and then used grub-install to install the grub onto the efi partition.

jglathe, 2024-11-05:
> We used cp to copy the ubuntu file system from the USB disk to the SSD disk, and then used grub-install to install the grub onto the efi partition.

Ok, nice. But I'm afraid you would need to edit /etc/fstab to get this one up then. This is the partition view of the bootable image (after using it on a stick, though):

[image]

the /etc/fstab says:
```
jglathe@x14-jg:~$ cat /media/jglathe/rootfs612/etc/fstab 
LABEL=rootfs612	/	ext4	discard	0	1
LABEL=BOOT612	/boot/efi	vfat	defaults	0	1

```
So, the partitions on the internal SSD would need these labels to be found when booting Linux from the SSD. You can choose different methods, like UUID, but label has proven to be quite robust.

jglathe, 2024-11-05:
Another curious question: Does your SnapDragon Dev Kit, per chance, have a HDMI out? This would be mdss-dp3, which is not in the device tree. But it would probably be easier to bring up than the DP Altmode nightmare we have to deal with now.

closed 2024-11-18 by jglathe