jglathe/linux_ms_dev_kit #27: WDK23 Periodic Reboot Issue

adilahmad17, 2024-11-07:
Hi! First off, I love and appreciate all the work that has been done to get Linux to boot up on these ARM machines. :) 

I'm using one of the provided pre-installed Linux images (6.8.8+). Under heavy load, the machine will periodically reboot. Is that something you also noticed? 

I played around with the power settings to avoid suspend, etc. but to no avail.  I initially thought this may be because of USB SSD heating/instability issues, so I installed the image onto the machine's SSD (alongside Windows). The issue appears again. I also tried installing the newer 6.10 kernel, but the issue persists on that kernel too. This issue does not happen on Windows, so I'm guessing its not a hardware issue. Any help is much appreciated!

jglathe, 2024-11-07:
Hi Adil, from experience I would say those kernels are not very new :) Anyway, what you describe may be a thermal alarm (you said heavy load) that is not appropriately handled. To be frank, I have not had thermal-related crashes for about 9 months(?) now. 
How does the cooling sound? Is it continuous fan, gradually getting louder with temperature of the box? Or is it "emergency blasts" every 2 minutes or so? If it's the latter, then I recommend booting the box with Windows, to reset the EFI variable (and some EC stuff apparently) to get the box back into "normal" thermal management mode. The thermal management mode appears to get broken by failed boots and unclean shutdowns. And sometimes it might happen out of the blue, but I haven't seen it happen every time.
As for a stable kernel, 6.11 might be a good one. But aside from some not-really-shutting-down issue with the 6.8 ones, it should be good. The newest sorta general-use image would be Ubuntu_Desktop_24.04_x1e_6.11rc_v7.img.xz as of now. 
The WDK dtb is in the phase of [upstreaming](https://lore.kernel.org/all/20241107-jg-blackrock-for-upstream-v8-0-0305621f17f2@oldschoolsolutions.biz/), hopefully it will be included in upstream kernels eventually.
Also trying to get this into [Ubuntu Concept X1E](https://discourse.ubuntu.com/t/ubuntu-24-10-concept-snapdragon-x-elite/48800) which also should boot sc8280xp hardware.

adilahmad17, 2024-11-08:
Thanks for the quick response and suggestions! (_new-ish_ kernels then ;)) 

It seemed like a thermal alarm to me as well, since it happened both times when I was building the "-el2" kernel using all cores. On a side-note, I love how well kvm and protected-kvm (with the slbounce trick) work using the "-el2" kernels (based on my past experience). 

Regarding the cooling sounds, now that I think about it, I've barely heard the fans kick in (which is odd since I recall the fans being somewhat loud previously on this machine when they do kick in). I did turn on the kit after several months, so I didn't think of that right away.
 
I'll try the EFI variable reset. I did a bunch of unclean shutdowns in the last day, so it could definitely be a culprit.  I'll also try the newer desktop image, if not. I'll report my findings here for posterity. Thanks again!

jglathe, 2024-11-08:
Interesting. Many users want kvm. I'm running 6.12-rc5-el2-x1e with no issue, [which is available as packages](https://drive.google.com/drive/folders/1Lps5o3FXroAJFDiKj18vutJbC1uld49s?usp=sharing).
Some observations on the el2 box:
- the UEFI BIOS is sometimes erratic in its choice of boot drive (when USB is connected too).
- thermal management behaves the same
- adsp and cdsp firmware are not loaded on EL2, needs some hacking to get this (although it should be possible)
- gpu acceleration is there without the ZAP shader firmware
- using a btrfs based system there, had to put it onto an external USB SSD instead of the internal nvme. Btrfs sporadically went into read-only state which is not good for a VM host. The root cause seems to be some sort of sensitivity of btrfs to pcie hardware errors (even when corrected and only reported). The SN740 1TB I had in the wdk seems to be some sensitive candidate. Working from an external 2TB SSD (SATA-III, but fast enough) solved this for good. 
- trying to enable nested virt on sc8280xp is impossible (cpu core does not support it), but it is being worked on for x1e80100. Which is another level of cool.

adilahmad17, 2024-11-08:
Happy to report that the reboot issues are fully resolved by installing the Ubuntu_Desktop_24.04_x1e_6.11rc_v7.img.xz! In this version, I hear the fans ramp up (every 20-30s) on heavy load (e.g., kernel compilation). Could be some weird fan management issues on my particular dev kit that affects the older kernel images.

For the el2 box,

- I also noticed the UEFI BIOS behaving weirdly for the boot drive with USB connected (I guess it depends on the port to which it is connected and quite erratic in its selection of "fsX"). Installing the kernel on the SSD generally dissipates that issue though, and my `startup.nsh` script seems to boot and bounce reliably (for now). I'm using an ext4 root.

- Thanks for the suggestions on btrfs, firmware, and gpu acceleration! I'm mostly interested in gpu acceleration and I was able to enable acceleration.

- Nested virtualization on the x1e would be extremely cool! I haven't tested the virtualization capabilities on that CPU yet, but I will soon. Glad to have one working machine first. :) 

Again, thanks for all your help! I'm closing this issue.

closed 2024-11-08 by adilahmad17

jglathe, 2024-11-09:
This fast ramping up means its still in emergency blast mode. Booting to Windows is the only known to me to get this resolved. The history of discovery of this is [here](https://github.com/jglathe/linux_ms_dev_kit/discussions/9#discussioncomment-8069742). The blasts are tripping the core temperatures (or PMIC temperatures) over 85°C, it goes full blast until core temperatures fall below 60°C, and the fan idles again. One important detail I've seen on re-reading it: You need to make sure that the EFI partition gets mounted correctly on boot. It needs to be mounted (without a warning) on `/boot/efi`.
[image]
If there's a warning on the EFI system partition, it is recommended to install the following additional tools:
```
sudo ap-get install mtools dosfstools
```
On the next reboot (after booting to Windows) it should be mounted as intended. This should also preserve the state of the thermal management and reduce the chance it gets broken since the EFI partition is mounted. The newer kernels usually also support qseecom for mounting efivarfs.
```
jglathe@volterra:~/src/linux_ms_dev_kit$ mount|grep efi
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
/dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro)
```

adilahmad17, 2024-11-11:
Thanks! I did boot into Windows and ensure that my EFI partition is correctly mounted. The fan problem seems to be solved now. The kit has been running non-stop for the last couple of days without any issues! :)