jglathe/linux_ms_dev_kit #30: The PCIe x16 slot on the X Elite motherboard is disabled.

allokey, 2025-01-31:
[Bringing up the SnapDragon Dev Kit for Windows with Linux](https://github.com/jglathe/linux_ms_dev_kit/wiki/Bringing-up-the-SnapDragon-Dev-Kit-for-Windows-with-Linux-%E2%80%90-*with*-working-display)
I connected an NVIDIA 4090 via the onboard PCIe x16 slot and an external power supply, but the PCIe x8 lanes are not activated, preventing GPU detection. Additionally, the graphics driver is incompatible with the current kernel. Are there any solutions to these issues?

jglathe, 2025-01-31:
AFAIK the PCIe slot in the dev kit (clearly visible in the layout) is not wired up. I honestly don't know which PCIe port that would be, it will be missing in the x1e80100.dtsi, too.

allokey, 2025-01-31:
[image]
PCIe and HDMI have always been part of the design. They were temporarily omitted from production due to manufacturing process constraints (PCIe) and lack of hot-plug capability (HDMI), though in practice, these components can be manually surface-mounted onto the board.

jglathe, 2025-01-31:
This is nice and interesting info. However, does this slot work on Windows? Would be the first thing to check. From there on, there is still the issue of the pcie port being defined in the dtb and in the x1e80100.dtsi.

jglathe, 2025-01-31:
The sources are open (except maybe for the 4090). Current top of branch is [jg/6.13-7](https://github.com/jglathe/linux_ms_dev_kit/tree/jg/ubuntu-qcom-x1e-6.13). The kernel packages I build and publish are [here](https://drive.google.com/drive/folders/1Lps5o3FXroAJFDiKj18vutJbC1uld49s). The config I use is in `arch/arm64/config/ubuntu_x1e_defconfig`. And in the annotations.

allokey, 2025-02-01:
> This is nice and interesting info. However, does this slot work on Windows? Would be the first thing to check. From there on, there is still the issue of the pcie port being defined in the dtb and in the x1e80100.dtsi.

Yes, PCIe gen4 x8 and HDMI are available under Windows system. Also, I saw information about a '[V7 patch](https://lore.kernel.org/lkml/20241017030412.265000-8-quic_qianyu@quicinc.com/)'. However, NVIDIA does not have Windows on ARM GPU drivers.

jglathe, 2025-02-01:
Huh. Okay I checked, that patch appears to be applied and from the date (and a re-try to apply it) it is already in my tree. I have defined pcie3 in the DevKit dtb, however I would expect to need some control gpios for enabling it. I also defined mdss_dp3 for the hdmi port. Usually you need a connector definition, too, but its worth a try. Regarding nVidia drivers, I will take a look at the config, maybe somethings missing. Do you have other cards to check the slot? I will post the build and tree with the tag 6.13-8.

allokey, 2025-02-01:
Okey，"I saw an article in @geerlingguy blog about '[AMD Radeon PRO W7700 running on Raspberry Pi](https://www.jeffgeerling.com/blog/2024/amd-radeon-pro-w7700-running-on-raspberry-pi)'. I don't have an AMD graphics card here, but I can buy one, though shipping will take a few days. When I get the AMD graphics card, I will try it again.

jglathe, 2025-02-01:
First bootup with pcie3 enabled shows som hardware not coming completely up:

[image]

I will try and guess the reset and enable GPIOs A good data source would be better, though.

The HDMI port might work now, it is mdss_dp3. 

[x1e001de-devkit.dtb.txt](https://github.com/user-attachments/files/18630117/x1e001de-devkit.dtb.txt)

I will upload 6.13-8 when I have the missing GPIOs.

Mis012, 2025-02-02:
@jglathe: my guess would be
```
	perst-gpios = <&tlmm 143 GPIO_ACTIVE_LOW>;
	wake-gpios = <&tlmm 145 GPIO_ACTIVE_LOW>;
```

jglathe, 2025-02-02:
Oh I got lucky: https://lore.kernel.org/linux-arm-msm/20241129-topic-sl7_feat2-v2-2-fb6cf5660cfc@oss.qualcomm.com/

So that's settled. But, with pcie3 and mdss_dp3 enabled, I get errors (don't have the hardware attached):

- pcie3 phy doesn't come up
- mdss_dp3 throws errors

This could be because the hardware isn't there, maybe @allokey knows a little more. Anyway, to have both cases covered, I have added a separate dts file for the fully populated variant. I will upload the latest build (still 6.13-8) later on.

allokey, 2025-02-03:
Alright, let me try it out.

jglathe, 2025-02-03:
[x1e001de-devkit_pcie3_hdmi.dtb.txt](https://github.com/user-attachments/files/18638857/x1e001de-devkit_pcie3_hdmi.dtb.txt)

The build is [up](https://drive.google.com/drive/folders/1vAv4BRsX5lFBrAq_KmmysznR6pgHsGWy?usp=sharing), [source](https://github.com/jglathe/linux_ms_dev_kit/tree/6.13-8) too.

allokey, 2025-02-03:
> [x1e001de-devkit_pcie3_hdmi.dtb.txt](https://github.com/user-attachments/files/18638857/x1e001de-devkit_pcie3_hdmi.dtb.txt)
>
> The build is [...]

OK. I'll try installing it first, let's start with NVIDIA.

allokey, 2025-02-03:
Here is the English translation:

This [boot1.txt](https://github.com/user-attachments/files/18639901/boot1.txt) contains the complete boot log. I will proceed as follows:

1. Check the connection between the PC power supply and the graphics card.

2. Since HDMI is not functioning, I need to inspect the firmware.

jglathe, 2025-02-05:
Hi @allokey , thank you for the extensive log. Activation doesn't look right, looks to me like we're missing some regulators to get the phy's up. Maybe a look into the Surface pro (11?) might help, since it has the SD card reader connected via PCIe3. Although Microsoft is so odd with hardware... worth a try.