jglathe/linux_ms_dev_kit #31: Install to SSD?

BatMahn, 2025-02-01:
Is installing to SSD possible? Even if we preserve some of the partitions for the firmware bits?

jglathe, 2025-02-01:
Hi @BatMahn, yes it is. Of course it is a "know what you're doing" class excercise. What we can't do yet is to boot the Ubuntu Concept X1E image and just install to a disk. But you actualy can copy over the image to the SSD and get it to boot from there. 

[image]

As you can see, I preserved all the existing partitions and added a new one for the linux rootfs. For this I replaced the existing SSD with a 2TB model and dd'ed the contents of the original SSD over. The other GPT partitions 1-12 are home of the boot menu and UEFI, therefore I decided to keep them. Partition 13 is our ESP, 14 and 15 Windows RE and Windows root, #16 is our new filesystem. 

The general steps are: 

- make a complete bitwise backup of your SSD
- boot from the image
- copy over the rootfs partition of the image to the SSD
- mount and chroot the ESP and linux root on the SSD
- edit /etc/fstab
- install grub on the SSD
- umount
- reboot to SSD.

The devil is in the details, I would need to do a detailed wiki for this. I still have the hope to enable the Ubuntu Concept ISO to install on the Snapdragon Dev Kit.

BatMahn, 2025-02-02:
Appreciate the steps. I'll give it a go!

BatMahn, 2025-02-09:
Happy to report that the above steps seemed to do the trick!

I was stuck with the stock 512GB SSD, so I nuked the Windows Recovery partition plus main partition (insert Austin Powers Meme here about living dangerously).

The first boot wasn't happy, giving me a Windows recovery screen saying my PC needed to be repaired and it couldn't find the files to do so. A subsequent boot using the USB key again got me into the desktop where it had mounted my /dev/nvme0n1p14 as the root partition. So a quick install-grub/update grub did what it needed to do. Now I can boot with no USB key.

Cheers!

jglathe, 2025-02-09:
Congrats! Yes, well... too sad about Windows, it was useful. The fan profile thing is amazing, right? I mean I can get the box to scream like a jet at takeoff, but I need to do things like build the kernel. Otherwise its mostly silent, as it should be. If you need the original SSD contents like it was shipped, I actually saved it.

closed 2025-04-06 by jglathe