jglathe/linux_ms_dev_kit #34: Windows Dev Kit 2023 DP Support

dsx724, 2025-06-02:
```
[    0.050657] /soc@0/phy@88eb000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae90000
[    0.050822] /soc@0/phy@8903000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae98000
[    0.051243] /soc@0/display-subsystem@ae00000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.051318] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000
[    0.051372] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae98000
[    0.051425] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae90000
[    0.051455] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.051484] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.051631] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/display-controller@ae01000
[    0.051648] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/phy@88eb000
[    0.051682] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.051713] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.051857] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/display-controller@ae01000
[    0.051874] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/phy@8903000
[    0.051908] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.051938] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.052077] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/display-controller@ae01000
[    0.052132] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/phy@aec2a00
[    0.052176] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.052229] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.055988] /soc@0/phy@88eb000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae90000
[    0.056338] /soc@0/phy@8903000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae98000
[    0.057961] /soc@0/display-subsystem@ae00000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.058075] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000
[    0.058152] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae98000
[    0.058228] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae90000
[    0.058258] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.058290] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.058507] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/display-controller@ae01000
[    0.058526] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/phy@88eb000
[    0.058578] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.058610] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.058820] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/display-controller@ae01000
[    0.058838] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/phy@8903000
[    0.058887] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.058918] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.059123] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/display-controller@ae01000
[    0.059184] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/phy@aec2a00
[    0.059229] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000
[    0.059288] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.059444] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/phy@aec2a00
[    0.059755] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.059807] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.059844] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.059876] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.059932] /soc@0/display-subsystem@ae00000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    0.065412] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /connector
[    0.065771] /connector: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000
[    2.543212] platform ae00000.display-subsystem: Adding to iommu group 4
[    2.560828] /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    2.561967] /soc@0/display-subsystem@ae00000/displayport-controller@ae98000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    2.563109] /soc@0/display-subsystem@ae00000/displayport-controller@ae90000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    2.564369] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/clock-controller@af00000
[    2.742405] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae9a000
[    2.745122] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae98000
[    2.750558] /soc@0/display-subsystem@ae00000/display-controller@ae01000: Fixed dependency cycle(s) with /soc@0/display-subsystem@ae00000/displayport-controller@ae90000
[    2.860140] msm_dpu ae01000.display-controller: bound ae90000.displayport-controller (ops msm_dp_display_comp_ops [msm])
[    2.863628] msm_dpu ae01000.display-controller: bound ae98000.displayport-controller (ops msm_dp_display_comp_ops [msm])
[    2.868939] msm_dpu ae01000.display-controller: bound ae9a000.displayport-controller (ops msm_dp_display_comp_ops [msm])
[... cut ...]
[    2.922711] msm_dpu ae01000.display-controller: [drm:adreno_request_fw [msm]] loaded qcom/a660_sqe.fw from new location
[    2.923147] msm_dpu ae01000.display-controller: [drm:adreno_request_fw [msm]] loaded qcom/a660_gmu.bin from new location
[    2.969193] msm_dpu ae01000.display-controller: [drm] Cannot find any crtc or sizes
[    2.969599] msm_dpu ae01000.display-controller: [drm] Cannot find any crtc or sizes
[    2.969627] msm_dpu ae01000.display-controller: [drm] Cannot find any crtc or sizes
[    3.182525] msm_dpu ae01000.display-controller: [drm] fb0: msmdrmfb frame buffer device
```

jglathe, 2025-06-02:
I need to ask which one. MiniDP? Currently using it on my WDK2023. Looking fine here.

```
jglathe@volterra:~$ sudo cat /sys/devices/platform/soc@0/ae00000.display-subsystem/ae01000.display-controller/drm/card1/card1-DP-3/edid |edid-decode 
edid-decode (hex):

00 ff ff ff ff ff ff 00 06 b3 27 32 00 00 00 00
0d 21 01 04 b5 46 28 78 3f 59 95 af 4f 42 af 26
0f 50 54 bf ef 80 d1 c0 b3 00 a9 40 01 01 90 40
81 80 81 40 81 c0 4d d0 00 a0 f0 70 3e 80 30 20
35 00 ba 89 21 00 00 1a 00 00 00 ff 00 52 33 4c
4d 44 57 30 32 32 30 35 38 0a 00 00 00 fd 00 32
3c a0 a0 3c 01 0a 20 20 20 20 20 20 00 00 00 fc
00 56 41 33 32 55 51 53 42 0a 20 20 20 20 01 c0

02 03 2a f1 52 01 03 04 05 07 10 12 13 14 16 1f
20 21 22 5d 5e 60 61 23 09 07 07 83 01 00 00 e3
05 e3 01 e6 06 07 01 60 60 45 56 5e 00 a0 a0 a0
29 50 30 20 35 00 ba 89 21 00 00 1e a3 66 00 a0
f0 70 1f 80 30 20 35 00 ba 89 21 00 00 1a ab 22
a0 a0 50 84 1a 30 30 20 36 00 ba 89 21 00 00 1e
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a8

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.4
  Vendor & Product Identification:
    Manufacturer: AUS
    Model: 12839
    Made in: week 13 of 2023
  Basic Display Parameters & Features:
    Digital display
    Bits per primary color channel: 10
    DisplayPort interface
    Maximum image size: 70 cm x 40 cm
    Gamma: 2.20
    DPMS levels: Off
    Supported color formats: RGB 4:4:4, YCrCb 4:4:4, YCrCb 4:2:2
    Default (sRGB) color space is primary color space
    First detailed timing includes the native pixel format and preferred refresh rate
    Display is continuous frequency
  Color Characteristics:
    Red  : 0.6845, 0.3095
    Green: 0.2597, 0.6845
    Blue : 0.1503, 0.0595
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    Apple   :   640x480    66.666667 Hz   4:3     35.000 kHz     30.240000 MHz
    DMT 0x05:   640x480    72.808802 Hz   4:3     37.861 kHz     31.500000 MHz
    DMT 0x06:   640x480    75.000000 Hz   4:3     37.500 kHz     31.500000 MHz
    DMT 0x08:   800x600    56.250000 Hz   4:3     35.156 kHz     36.000000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0a:   800x600    72.187572 Hz   4:3     48.077 kHz     50.000000 MHz
    DMT 0x0b:   800x600    75.000000 Hz   4:3     46.875 kHz     49.500000 MHz
    Apple   :   832x624    74.551266 Hz   4:3     49.726 kHz     57.284000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x11:  1024x768    70.069359 Hz   4:3     56.476 kHz     75.000000 MHz
    DMT 0x12:  1024x768    75.028582 Hz   4:3     60.023 kHz     78.750000 MHz
    DMT 0x24:  1280x1024   75.024675 Hz   5:4     79.976 kHz    135.000000 MHz
    Apple   :  1152x870    75.061550 Hz 192:145   68.681 kHz    100.000000 MHz
  Standard Timings:
    DMT 0x52:  1920x1080   60.000000 Hz  16:9     67.500 kHz    148.500000 MHz
    DMT 0x3a:  1680x1050   59.954250 Hz  16:10    65.290 kHz    146.250000 MHz
    DMT 0x33:  1600x1200   60.000000 Hz   4:3     75.000 kHz    162.000000 MHz
    DMT 0x2a:  1400x1050   59.978442 Hz   4:3     65.317 kHz    121.750000 MHz
    DMT 0x23:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz
[... 55 lines cut ...]
  HDR Static Metadata Data Block:
    Electro optical transfer functions:
      Traditional gamma - SDR luminance range
      Traditional gamma - HDR luminance range
      SMPTE ST2084
    Supported static metadata descriptors:
      Static metadata type 1
    Desired content max luminance: 96 (400.000 cd/m^2)
    Desired content max frame-average luminance: 96 (400.000 cd/m^2)
    Desired content min luminance: 69 (0.293 cd/m^2)
  Detailed Timing Descriptors:
    DTD 2:  2560x1440   59.950550 Hz  16:9     88.787 kHz    241.500000 MHz (698 mm x 393 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   5 Vback   33 Vpol P
    DTD 3:  3840x2160   29.980602 Hz  16:9     65.688 kHz    262.750000 MHz (698 mm x 393 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   5 Vback   23 Vpol N
    DTD 4:  1440x900    59.901458 Hz   8:5     55.469 kHz     88.750000 MHz (698 mm x 393 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   6 Vback   17 Vpol P
Checksum: 0xa8
jglathe@volterra:~$ uname -a
Linux volterra 6.15.0-jg-4-qcom-x1e #4 SMP PREEMPT_DYNAMIC Fri May 30 21:11:10 CEST 2025 aarch64 aarch64 aarch64 GNU/Linux
```
This build carries debug code for dp altmode with 4 lanes. This works on the WDK, too, but only on usb1 (the most rear type-c port). USB0 doesn't give a video out, still debugging.

dsx724, 2025-06-02:
Yes, the MiniDP. This was working in Linux 6.13.2 which I switched back to. I have not tested the Type-C interfaces. I attempted to read the edid of all 3 and failed with 6.15.

jglathe, 2025-06-02:
There's both commands, edid-decode (which I used) and decode-edid (which works with a file). As I mentioned, sitting in front of it and its working great.

dsx724, 2025-06-02:
Ahh I'm not sure what I did different but it's reading now. I think some of the updates to HDMI handling behavior upstream broke the display.

```
edid-decode (hex):

00 ff ff ff ff ff ff 00 4c 2d 13 73 00 00 00 00
01 20 01 04 b5 5f 36 78 3b d4 15 ae 4f 42 b0 26
0b 50 54 21 88 00 81 c0 81 00 81 80 95 00 a9 c0
b3 00 01 01 01 01 4d d0 00 a0 f0 70 3e 80 30 20
35 00 ad 11 32 00 00 1a 00 00 00 fd 0c 30 90 45
45 8f 01 0a 20 20 20 20 20 20 00 00 00 fc 00 4f
64 79 73 73 65 79 20 47 37 30 4e 43 00 00 00 ff
00 48 31 41 4b 35 30 30 30 30 30 0a 20 20 02 3f

02 03 33 f0 47 61 10 3f 04 03 5f 76 23 09 07 07
83 01 00 00 e3 05 c3 01 6d 1a 00 00 02 0f 30 90
00 04 73 03 60 0f e6 06 05 01 73 61 00 e5 01 8b
84 90 59 56 5e 00 a0 a0 a0 29 50 30 20 35 00 ad
11 32 00 00 1a 6f c2 00 a0 a0 a0 55 50 30 20 35
00 ad 11 32 00 00 1a 04 74 80 18 71 38 2d 40 58
2c 45 00 ad 11 32 00 00 1e 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d8

70 12 79 03 00 03 01 64 0f d0 01 84 ff 0e 2f 02
af 80 57 00 6f 08 59 00 07 80 09 00 d5 2c 02 04
ff 0e 2f 02 f7 80 1f 00 6f 08 59 00 02 00 04 00
4c d0 00 04 ff 0e 9f 00 2f 80 1f 00 6f 08 3d 00
02 00 04 00 08 ec 00 04 ff 09 9f 00 2f 80 1f 00
9f 05 66 00 02 00 04 00 59 87 00 04 7f 07 9f 00
2f 80 1f 00 37 04 4c 00 02 00 04 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 9c 90

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.4
  Vendor & Product Identification:
    Manufacturer: SAM
    Model: 29459
    Made in: week 1 of 2022
  Basic Display Parameters & Features:
    Digital display
    Bits per primary color channel: 10
    DisplayPort interface
    Maximum image size: 95 cm x 54 cm
    Gamma: 2.20
    DPMS levels: Off
    Supported color formats: RGB 4:4:4, YCrCb 4:4:4, YCrCb 4:2:2
    First detailed timing includes the native pixel format and preferred refresh rate
    Display is continuous frequency
  Color Characteristics:
    Red  : 0.6826, 0.3095
    Green: 0.2587, 0.6875
    Blue : 0.1484, 0.0439
    White: 0.3134, 0.3291
  Established Timings I & II:
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0a:   800x600    72.187572 Hz   4:3     48.077 kHz     50.000000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
  Standard Timings:
    DMT 0x55:  1280x720    60.000000 Hz  16:9     45.000 kHz     74.250000 MHz
    DMT 0x1c:  1280x800    59.810326 Hz  16:10    49.702 kHz     83.500000 MHz
    DMT 0x23:  1280x1024   60.019740 Hz   5:4     63.981 kHz    108.000000 MHz
    DMT 0x2f:  1440x900    59.887445 Hz  16:10    55.935 kHz    106.500000 MHz
    DMT 0x53:  1600x900    60.000000 Hz  16:9     60.000 kHz    108.000000 MHz (RB)
    DMT 0x3a:  1680x1050   59.954250 Hz  16:10    65.290 kHz    146.250000 MHz
  Detailed Timing Descriptors:
    DTD 1:  3840x2160   59.996625 Hz  16:9    133.312 kHz    533.250000 MHz (941 mm x 529 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
[... 72 lines cut ...]

Block 2, DisplayID Extension Block:
  Version: 1.2
  Extension Count: 0
  Display Product Type: Standalone display device
  Video Timing Modes Type 1 - Detailed Timings Data Block:
    DTD:  3840x2160  120.000000 Hz  16:9    270.000 kHz   1188.000000 MHz (aspect 16:9, no 3D stereo, preferred)
               Hfront  176 Hsync  88 Hback  296 Hpol P
               Vfront    8 Vsync  10 Vback   72 Vpol P
    DTD:  3840x2160  143.989899 Hz  16:9    323.977 kHz   1425.500000 MHz (aspect 16:9, no 3D stereo)
               Hfront  248 Hsync  32 Hback  280 Hpol P
               Vfront    3 Vsync   5 Vback   82 Vpol N
    DTD:  3840x2160   59.996625 Hz  16:9    133.312 kHz    533.250000 MHz (aspect 16:9, no 3D stereo)
               Hfront   48 Hsync  32 Hback   80 Hpol P
               Vfront    3 Vsync   5 Vback   54 Vpol N
    DTD:  2560x1440  143.973257 Hz  16:9    222.151 kHz    604.250000 MHz (aspect 16:9, no 3D stereo)
               Hfront   48 Hsync  32 Hback   80 Hpol P
               Vfront    3 Vsync   5 Vback   95 Vpol N
    DTD:  1920x1080  143.981451 Hz  16:9    166.587 kHz    346.500000 MHz (aspect 16:9, no 3D stereo)
               Hfront   48 Hsync  32 Hback   80 Hpol P
               Vfront    3 Vsync   5 Vback   69 Vpol N
  Checksum: 0x9c
Checksum: 0x90

```

jglathe, 2025-06-03:
Counter-question. Can you boot with the kernel that I published? Like, [6.15.0-jg-5](https://drive.google.com/drive/folders/1Lps5o3FXroAJFDiKj18vutJbC1uld49s)? Also, with the newer ones, you can do `zcat /proc/config.gz` and see the config the kernel is built with.

alexcaoys, 2025-07-05:
> ```
> [   10.305251] msm_dpu ae01000.display-controller: direct-loading qcom/x1e80100/Thundercomm/DEVKIT/qcdxkmsuc8380.mbn
> [   10.330331] msm_dpu ae01000.display-controller: Loaded FW: qcom/x1e80100/Thundercomm/DEVKIT/qcdxkmsuc8380.mbn, sha256: ab17276c807c7cd4b187f4c0ec8de99ad3327dfa293622ea38f3e34df8ae3795
> ```

I am not sure if this is related. But recently when I am testing a Snapdragon Dev Kit, the display is not working on any ports, with ubuntu-qcom-x1e-6.16.0-rc4-jg-0. I also didn't see these two lines, or any lines with msm_dpu. 
I tried to use the kernel comes from your image (6.13rc), it works. 
I haven't tested any other kernel versions. But from reading logs and codes changed, I am assuming some new codes simply block msm_dpu and gpu to load.

jglathe, 2025-07-05:
Hmm using 6.16.0-rc4-jg-0 on the Snapdragon Dev Kit, working fine. When booting I sometimes need to re-plug the type-c to dp cable to get display. What are you using as display? a dmesg of the plug event might help to find out whats up, there is debug code in it.

There are a lot of differences between 6.13rc and 6.16rc - the dp altmode now supports 4 lanes, maybe that's the issue.

[image]