jglathe/linux_ms_dev_kit #35: Screen dead with 6.15.*

h-2, 2025-06-04:
Hey, here are the details about the non-working screen on 6.15.0-jg-2

First, two regressions that are likely unrelated, but maybe not?

1. Cold-starting the laptop without any peripherals attached will immediately hang the system after kernel selection with frozen grub menu).
2. The FDE password prompt does not appear, the internal screen is still off at this point. It is possible to enter the password blindly and the system boots.

These don't appear on 6.14.0-35-qcom-x1e appear, although I had them before...

Now the actual problem that the screen is not working. I have uploaded a full dmesg and dmesg "updates" when plugging the screen in and unplugging it here: http://hauswedell.net/dmesg.tar.gz

To me, it seems like nothing screen-related pops up, and it only detects the usb-hub and attached devices? However, the system does recognise the screen and thinks it is there and on with correct resolution (`swaymsg -t get_outputs`). I can move windows to the screen area and back...

jglathe, 2025-06-04:
Could you try 6.15.0-jg-5? Grub menu and FDE password are probably out of my league, but the latter may be a setup issue. I need to ask: we're talking about external screen? Would be good to know if you can read the EDID of the external screen. Like this:

`sudo cat /sys/devices/platform/soc@0/ae00000.display-subsystem/ae01000.display-controller/drm/card1/card1-DP-1/edid |edid-decode`

It could be DP-1 or DP-2, need to look what device names are there. Could even be a different address of the display controller, although working here on the Snapdragon DevKit.

h-2, 2025-06-05:
> Could you try 6.15.0-jg-5?

Interesting! So I had posted the dmesg of jg-5 already at the bottom of the first post, but I hadn't realised the behaviour is changed with regard to the bootsplash/FDE-prompt: it now works, and it also works on the external screen. Only when GDM loads, the external monitor now goes blank (same as before).

> I need to ask: we're talking about external screen?

Yes, it's about an external USB-C-attached screen. The internal screen has always worked (except being off during FDE-prompt on some kernels).

> Would be good to know if you can read the EDID of the external screen. Like this:

The exact command works:

```
# cat /sys/devices/platform/soc@0/ae00000.display-subsystem/ae01000.display-controller/drm/card1/card1-DP-1/edid |edid-decode
edid-decode (hex):

00 ff ff ff ff ff ff 00 41 0c 48 09 89 00 00 00
0b 1f 01 04 b5 3c 22 78 3a 5f 75 ab 50 46 aa 27
0e 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40
81 c0 01 01 01 01 4d d0 00 a0 f0 70 3e 80 30 40
35 00 55 50 21 00 00 1a 00 00 00 ff 00 55 4b 38
32 31 31 31 30 30 30 31 33 37 00 00 00 fc 00 50
48 4c 20 32 37 39 50 31 0a 20 20 20 00 00 00 fd
00 17 4b 1e 8c 3c 01 0a 20 20 20 20 20 20 01 00

02 03 18 f1 4b 01 03 05 14 04 13 1f 12 02 11 90
23 09 07 07 83 01 00 00 a3 66 00 a0 f0 70 1f 80
30 20 35 00 55 50 21 00 00 1a 56 5e 00 a0 a0 a0
29 50 30 20 35 00 55 50 21 00 00 1e 02 3a 80 18
71 38 2d 40 58 2c 45 00 55 50 21 00 00 1e 4d 6c
80 a0 70 70 3e 80 30 20 3a 00 55 50 21 00 00 1a
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e6

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.4
  Vendor & Product Identification:
    Manufacturer: PHL
    Model: 2376
    Serial Number: 137
    Made in: week 11 of 2021
  Basic Display Parameters & Features:
    Digital display
    Bits per primary color channel: 10
    DisplayPort interface
    Maximum image size: 60 cm x 34 cm
    Gamma: 2.20
    DPMS levels: Off
    Supported color formats: RGB 4:4:4, YCrCb 4:4:4, YCrCb 4:2:2
    First detailed timing includes the native pixel format and preferred refresh rate
  Color Characteristics:
    Red  : 0.6689, 0.3134
    Green: 0.2763, 0.6669
    Blue : 0.1533, 0.0576
    White: 0.3134, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.081663 Hz   9:5     31.467 kHz     28.320000 MHz
    DMT 0x04:   640x480    59.940476 Hz   4:3     31.469 kHz     25.175000 MHz
    Apple   :   640x480    66.666667 Hz   4:3     35.000 kHz     30.240000 MHz
    DMT 0x05:   640x480    72.808802 Hz   4:3     37.861 kHz     31.500000 MHz
    DMT 0x06:   640x480    75.000000 Hz   4:3     37.500 kHz     31.500000 MHz
    DMT 0x08:   800x600    56.250000 Hz   4:3     35.156 kHz     36.000000 MHz
    DMT 0x09:   800x600    60.316541 Hz   4:3     37.879 kHz     40.000000 MHz
    DMT 0x0a:   800x600    72.187572 Hz   4:3     48.077 kHz     50.000000 MHz
    DMT 0x0b:   800x600    75.000000 Hz   4:3     46.875 kHz     49.500000 MHz
    Apple   :   832x624    74.551266 Hz   4:3     49.726 kHz     57.284000 MHz
    DMT 0x10:  1024x768    60.003840 Hz   4:3     48.363 kHz     65.000000 MHz
    DMT 0x11:  1024x768    70.069359 Hz   4:3     56.476 kHz     75.000000 MHz
[... 98 lines cut ...]
    1024x768 @ 70.069 Hz
    1024x768 @ 60.004 Hz
    832x624 @ 74.551 Hz
    800x600 @ 75.000 Hz
    800x600 @ 72.188 Hz
    800x600 @ 60.317 Hz
    800x600 @ 56.250 Hz
    720x576 @ 50.000 Hz (16:9)
    720x576 @ 50.000 Hz (4:3)
    720x480 @ 60.000 Hz (16:9)
    720x480 @ 60.000 Hz (4:3)
    720x480 @ 59.940 Hz (16:9)
    720x480 @ 59.940 Hz (4:3)
    640x480 @ 75.000 Hz
    640x480 @ 72.809 Hz
    640x480 @ 66.667 Hz
    640x480 @ 60.000 Hz (4:3)
    640x480 @ 59.940 Hz
    640x480 @ 59.940 Hz (4:3)
    720x400 @ 70.082 Hz
```

The mouse and keyboard attached to the external monitor also work.

Thanks for your help!

jglathe, 2025-06-05:
> I have tried attaching a different screen via USB-C (LG Electronics LG ULTRAWIDE 307NTYTBR626) and that one works with the usual hiccups (sometimes needs 2-3 tries).

For that one the dmesg would be interesting. I usually open with `dmesg -W`, do the action I want to log, copy from the console / terminal.

Re Screen goes blank: That's something dífferent than "no signal". May I ask what's your setup? Which distro?

h-2, 2025-06-06:
Thank you for your help!

> For that one the dmesg would be interesting. I usually open with dmesg -W, do the action I want to log, copy from the console / terminal.

I have uploaded logs here:
https://hauswedell.net/dmesg_w_screenfail (Philips)
https://hauswedell.net/dmesg_w_screenworks (LG ULTRAWIDE)

> Re Screen goes blank: That's something dífferent than "no signal".

Sorry, I meant "no signal", i.e. the screen says that there is no signal and goes into powersaving.

> May I ask what's your setup? Which distro?

I am using the Ubuntu concept setup. I installed it some time last year and upgraded to 25.04 in April.

jglathe, 2025-06-22:
Did you try out 6.16.0-rc2-jg-1? It has quite stable support for dp altmode.  Running it here with virtually no issues.

jglathe, 2025-06-24:
Hmm I would need some data. which laptop model? T14s? Also, do you have a dmesg with the plug retries? Do you have tried both orientations? (of course you did)

h-2, 2025-06-24:
> Hmm I would need some data. which laptop model? T14s? Also, do you have a dmesg with the plug retries?

I posted a bunch of dmesgs earlier (see beginning of thread). Since the behaviour hasn't changes since 6.15, I assume the dmesg hasn't changed, or do you need a new one?

Yes, it's a T14s 64GB non-oled

> Do you have tried both orientations? (of course you did)

You mean the usb-c upside down? I tried that before but not recently.

h-2, 2025-06-24:
Ok, curiously, on the second USB-C port, after a couple of tries, I can get the system to turn on the external screen in 640x480 mode in about one of five attempts (the other also result in no-signal). This is what ocassionally happens on older kernel, as well, except that there the working mode is about as likely. I have tried unplugging and replugging on the second port, about 15 times now, and it never yields a working screen. I have also tried a bunch of time more on the primary usb-c-port, and it never results in 640x480 mode.

This is the log from a an attempt that yields 640x480:
```
[  614.303545] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.304381] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.304845] fda000.phy-mux: qmp_combo_mux_set() enter mode=1, altmode=0
[  614.304852] qcom-qmp-combo-phy fda000.phy: typec_mux_set: switching from qmpphy mode 0 to 2
[  614.307490] fda000.phy-mux: qmp_combo_mux_set() exit mode=1, new_mode=2, altmode=0
[  614.310660] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.310664] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.652817] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.652839] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.652849] fda000.phy-mux: qmp_combo_mux_set() enter mode=1, altmode=0
[  614.652856] qcom-qmp-combo-phy fda000.phy: typec_mux_set: same qmpphy mode, bail out
[  614.656327] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.656332] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.783814] usb 6-1: new high-speed USB device number 24 using xhci-hcd
[  614.839840] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.839860] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.840556] fda000.phy-mux: qmp_combo_mux_set() enter mode=5, altmode=0
[  614.840564] qcom-qmp-combo-phy fda000.phy: typec_mux_set: switching from qmpphy mode 2 to 0
[  614.843356] fda000.phy-mux: qmp_combo_mux_set() exit mode=5, new_mode=0, altmode=0
[  614.846198] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.846202] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  614.912329] usb 6-1: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=78.23
[  614.912338] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  614.912343] usb 6-1: Product: USB2.1 Hub
[  614.912348] usb 6-1: Manufacturer: GenesysLogic
[  614.954620] hub 6-1:1.0: USB hub found
[  614.955124] hub 6-1:1.0: 6 ports detected
[  615.085250] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.085264] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.085269] fda000.phy-mux: qmp_combo_mux_set() enter mode=5, altmode=0
[  615.085273] qcom-qmp-combo-phy fda000.phy: typec_mux_set: same qmpphy mode, bail out
[  615.099537] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.099542] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.099544] fda000.phy-mux: qmp_combo_mux_set() enter mode=5, altmode=0
[  615.099545] qcom-qmp-combo-phy fda000.phy: typec_mux_set: same qmpphy mode, bail out
[  615.553972] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.553990] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.553999] fda000.phy-mux: qmp_combo_mux_set() enter mode=5, altmode=0
[  615.554006] qcom-qmp-combo-phy fda000.phy: typec_mux_set: same qmpphy mode, bail out
[  615.567081] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.567090] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=0, orientation=1
[  615.567095] fda000.phy-mux: qmp_combo_mux_set() enter mode=5, altmode=0
[  615.567099] qcom-qmp-combo-phy fda000.phy: typec_mux_set: same qmpphy mode, bail out
[  615.697579] [drm:msm_dp_bridge_get_modes [msm]] *ERROR* failed to get DP sink modes, rc=0
[  615.769908] [drm:msm_dp_bridge_get_modes [msm]] *ERROR* failed to get DP sink modes, rc=0
[  615.790799] usb 6-1: USB disconnect, device number 24
[  615.913141] usb 7-1: new SuperSpeed USB device number 14 using xhci-hcd
[  615.934142] usb 7-1: New USB device found, idVendor=05e3, idProduct=0625, bcdDevice=78.23
[  615.934163] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  615.934168] usb 7-1: Product: USB3.2 Hub
[  615.934173] usb 7-1: Manufacturer: GenesysLogic
[  615.996937] hub 7-1:1.0: USB hub found
[  615.997292] hub 7-1:1.0: 5 ports detected
[  616.022180] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=1, orientation=1
[  616.022187] fda000.phy-switch: qmp_combo_switch_set() enter new_orientation=1, altmode=1, orientation=1
[  616.022191] fda000.phy-mux: qmp_combo_mux_set() enter mode=5, altmode=1
[... cut ...]
[  618.771291] hid-generic 0003:195D:2029.0028: input,hidraw5: USB HID v1.11 Device [Gaming keyboard] on usb-xhci-hcd.4.auto-1.1.2/input1
[  619.376593] usb 6-1.3: New USB device found, idVendor=046d, idProduct=08e5, bcdDevice= 0.21
[  619.376613] usb 6-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=1
[  619.376619] usb 6-1.3: Product: HD Pro Webcam C920
[  619.376624] usb 6-1.3: SerialNumber: F486333F
[  619.434632] usb 6-1.3: Found UVC 1.00 device HD Pro Webcam C920 (046d:08e5)
```

jglathe, 2025-06-24:
The HDMI port is another matter. No solution for this yet.

h-2, 2025-06-25:
[image]

Amazing! That did it, even worked on first try without replugging 🎉 

> You can copy them over to the right place (wherever that is on Fedora) and reboot.

I am on Ubuntu-concept. I just copied the dtb to the boot folder for now!

> The HDMI port is another matter. No solution for this yet.

Not important for me atm; just thought I'd mention it for completeness.

One minor thing: the screen (internal and external) is off again during the FDE password prompt. I had this on older kernels, too. It was fixed on the ubuntu-concept 6.14.0 kernels. As mentioned above, it was NOT working on 6.15.0-jg-2 but it was working on 6.15.0-jg-5.

h-2, 2025-06-25:
Out of curiosity: Do you have an idea why it worked on older kernels if the frequency information was missing?

h-2, 2025-06-26:
OK, I think we can close this issue! Thank you for fixing this!