jglathe/linux_ms_dev_kit #46: backlight brightness arbitrarily limited

McFacePunch, 2025-08-09:
I'm working on an EL2 setup for an XPS 13 9345 with snapdragon. The rc6 and 7 kernels (others untested/checked) are built with `bl->max = 2047;` which arbitrarily limits brightness.

https://github.com/jglathe/linux_ms_dev_kit/blob/jg/ubuntu-qcom-x1e-6.16.0-rc6-jg-2/drivers/gpu/drm/display/drm_dp_helper.c#L4288

Removing this line or setting 65535 fixes the issue.

On your kernels, 6.16-rc6+ (at least if not earlier)
```
cat /sys/class/backlight/dp_aux_backlight/{max_brightness,brightness,actual_brightness}
2047
2047
2047

ls -la /dev/drm_dp_aux*
crw------- 1 root root 237, 0 Aug  8 22:09 /dev/drm_dp_aux0
crw------- 1 root root 237, 1 Aug  8 22:09 /dev/drm_dp_aux1
crw------- 1 root root 237, 2 Aug  8 22:09 /dev/drm_dp_aux2
```
On mainline 6.16.
```
cat /sys/class/backlight/dp_aux_backlight/{max_brightness,brightness,actual_brightness}
65535
38934
38934

ls -la /dev/drm_dp_aux*
crw------- 1 root root 237, 0 Aug  8 10:59 /dev/drm_dp_aux0
crw------- 1 root root 237, 1 Aug  8 10:59 /dev/drm_dp_aux1
crw------- 1 root root 237, 2 Aug  8 10:59 /dev/drm_dp_aux2
```

jglathe, 2025-08-09:
Yep that's the one: `be13439e1e851 HACK: workaround for t14s OLED backlight`. Will investigate.

McFacePunch, 2025-08-09:
I haven't dealt with the gpu driver code before but from a quick read it looks like this was in place of a detection for the time being. So I'm not sure if its a Quirk table thing or what the real solution is, otherwise I'd be happy to put in a PR. I can try to look this weekend and run some tests, or if you want something tested let me know. 

But side note, currently `6.16.0-rc6-jg-2-qcom-x1e` with slbounce EL2 (built with the force EL2 always flag) and a dtb overlay applied to the prior kernel is working well on XPS 13 9345. I've got /dev/kvm and everything. Fairly stable and usable, all the important things works.

jglathe, 2025-08-09:
Just building 6.16-el2-jg-2 for a test, with this patch removed. This is the tagged version: https://github.com/jglathe/linux_ms_dev_kit/tree/sg/ubuntu-qcom-x1e80100-6.16-el2-jg-2

Using the el2 branch may be beneficial; it has code to actually preserve the originally booted adsp-lite firmware from UEFI boot and use this in EL2. It's not the full-fat stack, but you get (on EL2):

- battery charging
- dp altmode support on type-c

You also seem to get a susceptibility for reboots when plugging a USB device when suspended. Can't pin it down yet.
Otherwise this is quite great, running the T14s 64GB OLED with it and its awesome.

jglathe, 2025-08-09:
Curious: Has the XPS 13 9345 64GB also that cutmem issue when not on EL2? I'm working ([tinkering](https://github.com/jglathe/T14s-hacks)) a bit with grub to add the required options.

jglathe, 2025-08-09:
These EL2 options started appearing with 6.16.0-rc6-jg-2. Separated them out to 6.16-el2-jg-? afterwards becauso of issues when on EL1 (now fixed).

jglathe, 2025-08-11:
The 64GB issue is a BIOS bug (actually Gunyah/arm-smmu), which is not yet fixed afaik.

closed 2025-09-09 by McFacePunch