jglathe/linux_ms_dev_kit #50: KVM and EL2 on XPS13

Paul-Mysten, 2025-08-31:
Figured Id create this to discuss the current state and possibilities for moving it forward. With SLbounce and current kernels, EL2 booting seems to work just fine, even KVM seems to boot up happily and recognize EL2 on all cores.

However, upon starting a VM with hardware acceleration, the entire machine halts immediately. I haven't even been able to get so much as a crash dump. So I'm looking for steps to try and debug this and maybe get hardware virtualization working.

I'll have to find my notes on exactly what I've tried but so far the problem is it halts and reboots too quickly for anything to give me valuable info. I've stopped just shy of kernel debugging with a remote machine due to time but plan on picking that up soon.

So any tips? Suggestions? Methodology? Know of any existing discussions or people working on this specific topic? Maybe an IRC or similar? 

Thanks

jglathe, 2025-09-01:
… on EL2.

Paul-Mysten, 2025-09-02:
... On a dell xp13 9345, it crashes... When running SLbounce, with full EL2 per dmesg, using `qemu-system-aarch64 -M virt -cpu host -smp 1 -m 1G -nographic -enable-kvm` leads to an instant kernel panic and reboot.

jglathe, 2025-09-02:
Did you add the [parameter](https://github.com/jglathe/linux_ms_dev_kit/issues/50#issuecomment-3241077840) in the kernel command line?

jglathe, 2025-09-05:
May I ask what kernel / distro you're using? I'm using either Ubuntu Concept 25.04 kernels, or my own, which have a slightly extended kernel config, but KVM support is on on both of them. Currently setting up lxd on the T14s.

Paul-Mysten, 2025-09-06:
Sure, Ubuntu 25.10 concept with a couple different 6.16 kernels from your branches and the default one. Also tried some of the upstream branches. No luck on any so far in KVM, but otherwise everything else is fairly stable, only a few crashes when doing things like resuming from sleep on power loss but I dont even remember what kernel did that, its not all of them.

I made sure to use the EL2 kernels as well, I verified the CPU cores are all on EL2 via dmesg, kvm init looks good, etc.

McFacePunch, 2025-09-09:
Same OP, so long story short I got to a point where I could prevent crashes with the following:

```
menuentry 'EL2' --class ubuntu --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-6.16.0-rc6-jg-2-qcom-x1e-advanced-ce06a3ba-4b1f-4d50-ab8e-cb82b2ed159f' {
        recordfail
        load_video
        gfxmode $linux_gfx_mode
        insmod gzio
        if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
        insmod part_gpt
        insmod ext2
        search --no-floppy --fs-uuid --set=root cbab1051-3824-4b26-a521-a2aa8e41a31d
        echo    'Loading Linux 6.16.0-rc6-jg-2-qcom-x1e ...'
        linux   /vmlinuz-6.16.0-rc6-jg-2-qcom-x1e root=/dev/mapper/ubuntu--vg-ubuntu--lv ro  clk_ignore_unused pd_ignore_unused cma=128M nosplash console=tty0 modprobe.blacklist=qcom_geni_serial crashkernel=2G-4G:320M,4G-32G:512M,32G-64G:1024M,64G-128G:2048M,128G-:4096M, id_aa64mmfr0.ecv=1
        echo    'Loading initial ramdisk ...'
        initrd  /initrd.img-6.16.0-rc6-jg-2-qcom-x1e
        echo    'Loading device tree blob...'
        devicetree      /x1e-el2.dtb
}
```
Now when I do the prior `qemu-system-aarch64 -M virt -cpu host -smp 1 -m 1G -nographic -enable-kvm` instead of crashing I get a near total lockup with moments of control. From this i was able to get some logs finally.

```dmesg
[  287.078052] geni_i2c 884000.i2c: Timeout abort_m_cmd
[  289.061158] power_supply qcom-battmgr-bat: driver failed to report `serial_number' property: -110
[  290.086155] power_supply qcom-battmgr-bat: driver failed to report `voltage_now' property: -110
[  291.110158] power_supply qcom-battmgr-bat: driver failed to report `status' property: -110
[  292.134114] power_supply qcom-battmgr-bat: driver failed to report `cycle_count' property: -110
[  293.158181] power_supply qcom-battmgr-bat: driver failed to report `status' property: -110
[  295.078201] power_supply qcom-battmgr-bat: driver failed to report `status' property: -110
[  296.101126] power_supply qcom-battmgr-bat: driver failed to report `energy_full_design' property: -110
[  298.025147] power_supply qcom-battmgr-bat: driver failed to report `voltage_max_design' property: -110
[  299.110144] [drm:dpu_encoder_frame_done_timeout:2715] [dpu error]enc38 frame done timeout
[  299.230175] [drm:dpu_encoder_frame_done_timeout:2715] [dpu error]enc38 frame done timeout
[  300.197224] power_supply qcom-battmgr-bat: driver failed to report `energy_now' property: -110
[  300.712115] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1: hangcheck detected gpu lockup rb 0!
[  300.712291] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     completed fence: 6880
[  300.712382] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     submitted fence: 6881
[  301.224128] geni_i2c 884000.i2c: Timeout abort_m_cmd
[  301.734175] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1: hangcheck detected gpu lockup rb 0!
[  301.734339] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     completed fence: 6880
[  301.734429] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     submitted fence: 6881
[  302.328168] power_supply qcom-battmgr-bat: driver failed to report `voltage_max_design' property: -110
[  302.696069] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1: hangcheck detected gpu lockup rb 0!
[  302.696233] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     completed fence: 6880
[  302.696323] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     submitted fence: 6881
[  303.334138] power_supply qcom-battmgr-bat: driver failed to report `status' property: -110
[  303.721131] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1: hangcheck detected gpu lockup rb 0!
[  303.721294] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     completed fence: 6880
[  303.721386] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     submitted fence: 6881
[  304.357070] power_supply qcom-battmgr-bat: driver failed to report `status' property: -110
[  304.743029] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1: hangcheck detected gpu lockup rb 0!
[  304.743192] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     completed fence: 6880
[  304.743283] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     submitted fence: 6881
[  305.385111] power_supply qcom-battmgr-bat: driver failed to report `temp' property: -110
[  305.701072] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1: hangcheck detected gpu lockup rb 0!
[  305.701235] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     completed fence: 6880
[  305.701327] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1:     submitted fence: 6881
[  306.408014] power_supply qcom-battmgr-bat: driver failed to report `cycle_count' property: -110
[  306.729016] msm_dpu ae01000.display-controller: [drm:hangcheck_handler [msm]] *ERROR* 67.5.12.1: hangcheck detected gpu lockup rb 0!
[... cut ...]
[  453.772740] [drm:dpu_crtc_frame_event_cb [msm]] *ERROR* crtc106 event 1 overflow
[  453.789446] [drm:dpu_crtc_frame_event_cb [msm]] *ERROR* crtc106 event 1 overflow
[  453.806116] [drm:dpu_crtc_frame_event_cb [msm]] *ERROR* crtc106 event 1 overflow
[  453.822823] [drm:dpu_crtc_frame_event_cb [msm]] *ERROR* crtc106 event 1 overflow
[  453.839491] [drm:dpu_crtc_frame_event_cb [msm]] *ERROR* crtc106 event 1 overflow
[  454.614117] workqueue: bpf_prog_free_deferred hogged CPU for >1000000us 5 times, consider switching to WQ_UNBOUND
```

Paul-Mysten, 2025-09-20:
What do you think would be some helpful logs to dump here? I didn't see much else of value looking but maybe another pair of eyes would be good.

Paul-Mysten, 2025-09-22:
I'm pretty sure I have the stock bios and firmware because of a warning from the firmware puller utility that some of the newer firmware doesn't work due to.... I forget the reason.

jglathe, 2026-02-23:
Well, others like @alexvinarskis have the XPS 9345 up on EL2 and use it that way. Does it still not work?

jglathe, 2026-02-27:
kvm-ok should work, it does here.

```
jglathe@t14s-jg:~$ kvm-ok
INFO: /dev/kvm exists
KVM acceleration can be used

```
I have this additional parameter in the kernel command line, though: `id_aa64mmfr0.ecv=1` Without it actually starting a vm would reset the machine.

on #2: Yes, I use this bootup script for the efi shell:

```
jglathe@t14s-jg:~$ cat /boot/efi/startup.nsh
FS13:
load qebspilaa64.efi
load slbounce.efi
\EFI\BOOT\grubaa64.efi
exit

```

The FS may be different on yours (FS9) but otherwise this should do it.